RoboJackets/robocup-software #184: Interface with MSP430G2 via MBED to control kicker

justbuchanan, 2015-01-16:
The MSP430G2 will be the controller for the kicker board.  We will interface with it from the MBED over I2C.  We need to setup a "protocol" for the commands to make this happen.

Use Code Computer Studio to work with it.

How do you program it? / what is spy-bi-wire?

blueintegral, 2015-01-16:
You'll need a programmer like the MSP-FET430UIF. Spy-by-wire is TI's 2 port version of JTAG, which is what the programmer speaks.

jjones646, 2015-01-26:
Changing paths a bit and going with one of [tinyAVR]() MCUs for it's *much* simpler flashing interface. We wish to create a streamlined way to update the device's flash memory through the mbed, and progress with the Spy-Bi-Wire protocol was getting nowhere. tinyAVR microcontrollers have much more documentation online for programming over standard [SPI](http://en.wikipedia.org/wiki/Serial_Peripheral_Interface_Bus).

We are currently working with some [ATtiny13A](http://www.atmel.com/devices/ATTINY13A.aspx) MCUs that were laying around, and have already managed to make good headway on the flashing implementation. There is already [code for programming an AVR MCU](http://developer.mbed.org/cookbook/mAVRISP) on the mbed website. The development environment is easily ported for use with the [Arduino IDE](http://arduino.cc/en/main/software), and it should make it easier for non-software people to follow along with the kicker's logic.

Potential MCUs for final design are the [ATtiny85](http://www.digikey.com/product-detail/en/ATTINY85-20SUR/ATTINY85-20SURCT-ND/3789340) and [ATtiny441](http://www.digikey.com/product-detail/en/ATTINY441-SSUR/ATTINY441-SSURCT-ND/4437439). All [tinyAVR MCUs](http://www.atmel.com/products/microcontrollers/avr/tinyavr.aspx) use the same method for reading/writing their Flash & EEPROM memory.

justbuchanan, 2015-02-02:
Here's a good tutorial on how to read/write fuse bits with avrdude.  The fuse bits are basically settings for the chip.

http://web.engr.oregonstate.edu/~traylor/ece473/lectures/fuses.pdf

justbuchanan, 2015-02-03:
I got most of this stuff setup on the `kicker-attiny` branch.

Run `make kicker2015-prog` to compile everything in the `firmware/robot2015/kicker-board` directory into the `run/kickerFW` binary-formatted executable and copy it onto any plugged-in MBEDs.  Next time the MBED restarts, it will attempt to load this program to a connected ATtiny13 over SPI.  Right now it's set to verbose, so you can read status updates using `screen`.

TODO:
- [ ] Fix the AVR910 verify() function - it's broken right now and causing issues.
- [ ] Add the avr dev tools package to `ubuntu-packages.txt`

justbuchanan, 2015-02-03:
Also, in `main()` it's setup right now to compare the `kickerFW` file to the flash memory of the avr chip and only flashes it if they're different.  Right now the `verify()` function is broken, so it always reflashes the program.

justbuchanan, 2016-01-13:
@jjones646, can you confirm that the kicker board is using the ATtiny 84A?

jpfeltracco, 2016-02-07:
@justbuchanan For compiling kicker board code to run on the ATtiny 84a, the only thing that would need to change is the CMakeLists.txt file in the src-kckr directory right? I'm only changing:
`set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -O2 -mmcu=attiny44a -DF_CPU=9600000")`
to
`set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -O2 -mmcu=attiny84a -DF_CPU=8000000")`

I think I'm uploading code to the attiny correctly, but I'm not sure if what's being uploaded is correct.

jjones646, 2016-02-07:
@jpfeltracco What you have there is correct.

justbuchanan, 2016-02-25:
@jpfeltracco, it took quite a while to debug, but I finally got this all working!

The main issue was that the `writeFlashMemoryPage()` in `AVR910` was shifting the page number incorrectly before sending it to the avr chip.  Writing page zero worked fine with the incorrect implementation because shifting zero gives you zero.  The error caused page 1 of the binary file to actually be written to the avr's page 2.  Likewise, page 2 was written to flash page 4, etc.  The BusPirate helped tremendously because I could read out the flash memory contents with it after writing to it with the MBED.  Once I noticed that the two-page binary I wrote to it showed up as the correct page 0, followed by a page of `0xFF` values, followed by the correct page 1, it wasn't too hard to track down.

This is all fixed in 1fbd9f7a1249db9d9b518e1758c785d8781809e5.

While I was at it, I revived the `KickerBoard` class I wrote a while ago - it was kinda buried and forgotten about, but it implemented basically everything you had in your test `main.cpp` file.  The latest work is on the [justin/avr-fixes branch](https://github.com/RoboJackets/robocup-software/commits/justin/avr-fixes).  If you merge that into yours, you should be good to go.

cc @ashaw596 @jjones646

jpfeltracco, 2016-02-26:
@justbuchanan Awesome! I merged it into my branch. I'm glad that's finally sorted out. I'll start working on the SPI communication now that we have a way to program the chip directly from the MBED.

ryanstrat, 2016-02-26:
@Jeremy, you should talk with Daniiel from the electrical team. He's been
working on API between and arduino and the mbed for a few weeks now.

jpfeltracco, 2016-02-26:
Yup, we've been in contact!

danbudanov, 2016-02-26:
Jeremy, I think I know what was wrong on the chip side, and expect to fix
it today. Will there be anybody in the shop around 6 today?

jjones646, 2016-04-04:
### Related Trello Cards
[Kicker Logic Specs.](https://trello.com/c/1XqeDtD4/157-kicker-board-requirements)
[SPI Prototyping](https://trello.com/c/rvMa3PoB/146-kicker-board-functionality-prototyping)
[SPI Arduino](https://trello.com/c/vcj7MJEu/154-mbed-arduino-spi-communication)
[Kicker Voltage Meas.](https://trello.com/c/UIJUXWG6/143-kicker-board-attiny-voltage-measurements)

justbuchanan, 2016-05-18:
@jpfeltracco @danbudanov how's progress on this?

jpfeltracco, 2016-05-18:
Pretty good, we've got the major functionality finished, as well as an implementation of our communication protocol implemented in a header file shared between the KickerBoard class and main.c for the ATTINY. The only thing I wasn't sure about was how exactly we wanted the debug buttons set up. Will you be around Thursday to take a quick look?

justbuchanan, 2016-05-18:
Sounds great!  Yeah I'll be around Thursday - what time are you thinking?

jpfeltracco, 2016-05-18:
Does 7:30 pm work?

justbuchanan, 2016-05-18:
Yeah that should work - I'll confirm with you on Thursday

justbuchanan, 2016-05-21:
@ryanstrat assembled a kicker board, so I think the hardware's mostly ready!

[image]

jpfeltracco, 2016-05-22:
Nice! I'll try getting software on it to do some testing on Tuesday!

danbudanov, 2016-05-28:
Great! What can I do to help? Are we having any issues with the kicker?

justbuchanan, 2016-05-28:
I just put you on here b/c GitHub added a feature to allow for multiple assignees and you were working on this before :).  I think this is mostly good unless @jpfeltracco needs you to work on anything.

closed 2016-06-24 by jjones646