RoboJackets/robocup-software #185: Setup I/O expander with MBED to drive status/error LEDs.

justbuchanan, 2015-01-16:


justbuchanan, 2015-01-21:
@jpfeltracco, I mentioned this earlier, but just wanted to put it here for future reference.

Right now to interface with pins, we do something like the following:
```
DigitalOut myLed(Pin1);
myLed = 1; # turn the led on
```

It'd be nice to have usage of pins be agnostic of whether or not it's on the I/O expander or a regular pin, so the api should be something like this:
```
IOExpander exp(Pin2, Pin3, Pin4); # setup I/O expander SPI/I2C (can't remember which one it uses)
IOExpanderDigitalOut myLed2(exp, IOExpanderPin1);
myLed2 = 1; # turn the led on
```

That way we specify the connections when setting up the pin, but later usage is the same whether it's on the expander or a regular pin.

guyfleeman, 2015-02-01:
@jpfeltracco do you have a link to the library you're using for the MCP24017? I'm looking at integrating it into cmake.

jpfeltracco, 2015-02-01:
Yup, here's the one I've been using: http://developer.mbed.org/users/Nurbol/code/MCP23017/

closed 2016-02-03 by justbuchanan