RoboJackets/robocup-software #186: Use MBED to interface with ball sense hardware

justbuchanan, 2015-01-16:
We need to pulse the emitter and read the detector to determine whether or not we have captured the ball.

justbuchanan, 2015-04-20:
@gtrosawa what are the differences between the ball sensor setup this year and our previous one?

justbuchanan, 2015-04-24:
@jjones646, @gtrosawa, We're looking at the ball sensor bar from last year and I'm not quite sure which side is which.  One "bulb" is red and one's clear - do you know which side is the detector?

jjones646, 2015-04-24:
- Purple/Black = Detector
- Clear = Emitter

justbuchanan, 2015-11-04:
@jjones646, @ryanstrat what's the status on ball sense boards?

jjones646, 2015-11-04:
@justbuchanan These are ~1 month away I believe. The pcb design is finished - just a matter of getting the order in and then waiting at that point.

ryanstrat, 2015-11-05:
​We ran into some issues panelizing the board that I think we've worked
through. They'll be ordered within the week with a 2.5 week turnaround.

Ryan Strat
ryan@str.at
(678) 517-3393
RoboJackets | RoboCup Project Manager
Georgia Institute of Technology

JNeiger, 2016-01-13:
Is there a schematic for the connection between the Detector/Emitter and the mbed?

Right now I have the emitter connected to p20 and ground. The detector is connected to p19 and the same ground.

With this setup, whenever the emitter has power going through it, the detector values are shifted a significant amount (~500 units), almost as if there is some sort of cross talk between them.

justbuchanan, 2016-01-13:
@jjones646, any ideas what might be causing the shift in values he's seeing here?  I looked at the breadboard setup for this yesterday and it all looked right to me.

@JNeiger, can you post a link to the branch that your code is on so I can check it out?

JNeiger, 2016-01-13:
@justbuchanan https://github.com/JNeiger/robocup-software/tree/ball_sensor

justbuchanan, 2016-01-13:
I browsed through your code and it all looks good to me.  Can you also post a pic of your breakout board?

justbuchanan, 2016-01-13:
I think the solution may be to put a resistor in series with the photoresistor, as shown here: [image].

I forgot about having that resistor in there before, but that should fix the issue.  @jjones646, do you know what resistor the final board will have?

commit a43053c1 (2016-02-17): Added basic ball sensing functionality.

Closes #186
closed 2016-02-17 by justbuchanan