RoboJackets/robocup-software #187: Interface with mechanical base ID chip over I2C

justbuchanan, 2015-01-16:
This is how we will uniquely identify mechanical bases in order to do per-robot calibration.

jjones646, 2015-04-06:
@justbuchanan We have some of these now.

ra4king, 2015-04-15:
http://www.digikey.com/product-detail/en/DS2411R%2BT%26R/DS2411R%2BCT-ND/2045729

justbuchanan, 2015-04-15:
This might be helpful: https://github.com/contiki-os/contiki/blob/master/dev/ds2411/ds2411.c.

It's not written for the MBED, but shouldn't be hard to port.

ra4king, 2015-04-15:
That's the most cryptic code I've ever seen. I had to hunt down through multitudes of header files to dig for half of those variables/functions!

ra4king, 2015-04-17:
Got it working.
https://github.com/ra4king/robocup-software/commit/63a8d9287d61e1212b25b6226b291d1c16fd8659

justbuchanan, 2015-04-17:
Nice, that's awesome! :boom:.  The next step from there will be packaging it up into its own Ds2411 class.

ra4king, 2015-04-18:
#281

closed 2015-04-19 by justbuchanan