RoboJackets/robocup-software #204: Setup Ninja build system

justbuchanan, 2015-02-04:
CMake is sort of an intermediate build system in that it doesn't build code directly, but outputs makefiles or other build files that can then be run to build the code.  CMake uses the `make` "Generator" by default, but you can choose other ones by passing it a value using the "-G" flag.  You can see more about CMake generators on their documentation page: http://www.cmake.org/cmake/help/v3.0/manual/cmake-generators.7.html.

Ninja is a build system sort of similar to `make`, but does a better job of parallelizing the build and will compile the project much faster.  I tried Ninja on my computer and it ran almost twice as fast.  This isn't exactly a fair comparison because I ran it against a serial make run, which can be parallelized by passing the "-j" (jobs) flag to make.  Ninja still benchmarks faster overall though because it's a newer system and designed from the ground up for speed and parallelization.

Switching to Ninja on my system (Arch linux) was pretty easy and just involved replacing "make" in our makefile with "ninja" and adding "-GNinja" to all of our CMake calls.  I ran into some trouble when trying this on Ubuntu though, so I can't put this onto master yet.  If someone can figure this out and get it to work reliably on Ubuntu, it'll give us faster builds overall.

ehuang3, 2015-02-04:
Tried ccache?

justbuchanan, 2015-02-04:
Hmm.. haven't tried it before.  CMake only recompiles files that change or have #includes that change.  Does this do something different?  It says on their site that they check for compilations that are being done again and shortcuts them, but I don't know how that would do better than straight up CMake.

ehuang3, 2015-02-04:
If you do a make clean, CMake will recompile everything. If you are switching back and forth between branches, CMake will recompile the changed files again. If you switch between Debug mode and Release mode, expect to recompile, again. Those are the most common use cases for myself, personally, that ccache will take care of.

Overall, ccache is very easy to setup and is transparent to the build system.

justbuchanan, 2015-02-04:
Ok, cool, thanks for the suggestion!  I like the use case of switching back and forth between branches, which I do pretty often.  I'll see about adding that in.

justbuchanan, 2015-02-04:
Got some pretty awesome results with ccache, I'll be using this from now on! :boom:

ashaw596, 2015-02-04:
I tried ninja it works mostly for ubuntu. Approximately the same success rate as with the j flag on make. I think there's some race issue that occurs because of dependencies or something. Running it twice works!

justbuchanan, 2015-02-04:
Interesting...  It's possible that we don't have our dependencies setup quite right, so if we fix that it should work in one try.

commit 871cef14 (2015-08-24): switched from make to ninja. closes #204
commit 63c3caeb (2015-08-24): switched from make to ninja. closes #204
commit 0244cd6c (2015-08-25): switched from make to ninja. closes #204
commit 01ae4f0b (2015-10-07): Merge pull request #386 from RoboJackets/ninja

switched from make to ninja. closes #204
closed 2015-10-07 by justbuchanan