RoboJackets/robocup-software #214: Add git version info to robot firmware

justbuchanan, 2015-02-17:
In order to help with debugging, it's helpful to know what version of the firmware our robot is running.  To do this, we'll use the git SHA as a version id so that we can resolve the version on the robot to a specific commit in our codebase.

This would go well in either a `status` or `version` command in the robot's console/terminal.

Right now we have CMake setup to automatically generate a `git_version.cpp` file that has two variables: `git_version_hash` and `git_version_dirty`.  We'll need to compile this cpp file into the firmware binary and add a command to the terminal.

jgkamat, 2015-02-17:
I think I can take this one during the meeting today

justbuchanan, 2015-02-17:
Sounds good!  I j ust assigned it to you.

PR #221 (2015-02-22): Added info command to mbed firmware
Fixes #214 
I added the current SHA, the commit date/time, and the author, as well as a serial number (not so sure about this one).

It would be nice if someone could double check all my work with cmake, because that could break all the build scripts if i've messed up.
closed 2015-02-22 by guyfleeman