RoboJackets/robocup-software #217: Run automated soccer matches and report the results

justbuchanan, 2015-02-18:
This is a super long-term goal, but it's a good one.  It would be awesome to have a script that runs two instances of soccer against each other (optinally with a headless mode.  i.e. no gui) in a simulator match (with a specified time limit and score limit) and have it write the results to a file (maybe just the regular log file).

This would be a great way to automate some machine learning so that we could run a match with a specific config file, then tweak the config file based on the results and run more matches.  The hope would be that at the end of many of these automated matches, certain config parameters would be more optimized and ready for battle.

This would go well in a docker container using the Dockerfile that @jgkamat setup to run soccer.

This is a long-term goal and I don't think we're near readly to tackle it yet, but it's something to keep in mind for the future.

jgkamat, 2015-02-18:
The docker image i've been working on is [at the docker hub](https://registry.hub.docker.com/u/jgkamat/robocup-software/).

You can run
```
docker run -ti \
       -e DISPLAY=$DISPLAY \
       -v /tmp/.X11-unix:/tmp/.X11-unix \
       jgkamat/robocup-software:latest
```
on a computer with docker installed to start a graphical version of soccer! This works on Debian, but it dosen't work on my arch machine (something to do with the x server).

Right now, the dockerfile is in [my personal repository](https://github.com/jgkamat/docker). If people want to work on it, maybe we could move it to robojackets (or robocup-software), and/or create a robojackets dockerhub account...

For reference, [this](http://fabiorehm.com/blog/2014/09/11/running-gui-apps-with-docker/) is where I got the code to run gui apps in docker.

jgkamat, 2017-07-03:
This is a dupe of #841 (despite this being the original issue, that one has more information, so I'm closing this one).

closed 2017-07-03 by jgkamat