RoboJackets/robocup-software #219: The testing/TwoSideAttack play doesn't work

justbuchanan, 2015-02-18:
Didn't look into the issue, but noticed that the two robots ignore the ball and just get in each other's way.

jgkamat, 2015-06-26:
I'm not exactly sure whats wrong with this, the play seems to work fine, but the evaluation seems to be messing up (The robot always passes to the robot on the right). Is that what was wrong with the play?

Also, unrelated, but I think simulator is segfaulting on start on the master branch...

justbuchanan, 2015-06-26:
Yeah I think you're right, I just ran it and everything seems fine.  I wonder if something else changed that accidentally fixed this.

Yeah, Albert actually emailed me about that just a little while ago too haha.  That was my bad, I goofed up some simulator "cleanup" a little while ago.  Just pushed a fix to master though, so pull from there and you'll be good to go.

closed 2015-06-26 by justbuchanan