RoboJackets/robocup-software #228: Improve Defender and SubmissiveDefender tactics to clear ball

justbuchanan, 2015-02-22:
Right now if a ball gets into the defense zone, the defenders just hover around it.  It would be much better if they had a "clear" state so they get the ball and kick it out of the way or pass it to another robot.
[image]

joshhting, 2015-05-25:
The defense zone is the area indicated by the green circle?

justbuchanan, 2015-05-25:
The defense zone is the actually the white portion.  The green circle is sort of a lazy approximation to the goal zone that the defenders position themselves on.

justbuchanan, 2015-07-26:
As of our changes during comp in China, defenders now clear the ball if it happens to land in their mouth.  However, if the ball is just a couple cm in front of them, they'll leave it there.  We still we want to be conservative about having a defender leave their post, but we should make the defenders clear the ball in more cases than they do currently.

ashaw596, 2015-07-29:
Part of this and though a more high level, but I think not too difficult to
achieve I think we could write some simple heuristic that accounts for our
distance vs nearest oponent robot distance and trajectories to calculate a
probability of successful capture

jcarn, 2016-04-26:
I'll look into this and see if I think I am up to the task

jgkamat, 2016-04-26:
Look into the SubmissiveDefender Play, I think you'll only need to modify that file.

jcarn, 2016-04-27:
When choosing to clear the ball or not, is the ratio between how quickly the defender can clear and how quickly the opponent can reach the ball a good metric?

jgkamat, 2016-04-27:
That probably would be the best metric, but it would be hard to find out (we don't know how fast their robots are). Maybe distance would be an easy way to do this for now. 

Also, if the ball is moving, we want to intercept the ball where it is going to be, not where it is. Capture (should?) do this for you.

jcarn, 2016-04-27:
I think it is important to take into account the case where they are moving at high speed towards the ball, so I'll look into a hybrid method since a case involving acceleration on their part would be hard to measure.

Gotcha, I will use that. Thanks

PR #714 (2016-06-22): Added a subbehavior to submissive defender to clear the ball.
Submissive Defenders will now clear the ball if it comes within a certain radius of the goal and they can reach the ball before an opponent. Closes #228
closed 2016-06-22 by chachmu