RoboJackets/robocup-software #231: Sometimes MBED doesn't start after 'make robot2015-prog'

guyfleeman, 2015-02-22:
Sometimes after a prog command the mbed does not start properly, led's 1 and 2 do not flicker. I suspect this has to do with setting interrupt priorities.

jgkamat, 2015-02-22:
I can confirm this, it seems to be much more prevalent on my Debian system than my arch system. I don't know why though...

This is the command the script runs to restart the mbed, it only works when there is firmware on the machine already.

```
sudo bash -c "echo -ne '\rreboot\r' > /dev/mbed"

```

To get a better idea of the problem, run screen on the mbed while running this command.

guyfleeman, 2015-02-23:
@jgkamat do you remember any specific conditions that could have led to the failures, specifically usb ops that could generate interrupts? I've tried adding delay between the firmware copy and the reboot command using ```sleep 1.0``` which seemed to improve the situation but didn't fix it all together.

jgkamat, 2015-02-23:
I remember that when the issue happened, if I screened into the mbed, (or cat /dev/mbed), I would get a ton of garbage output. It may have something to do with buffering?

guyfleeman, 2015-02-23:
@jgkamat Ah, now that you mention it that makes a lot of sense. It seems that the stdio buffer isn't always flushed prior to the to reboot. I wonder if this could cause complications with the copy? All USB traffic is routed through stdio I think.

When I screened into it it just failed? Did you get the garbage on arch? Maybe I should do my initial debugging there.

jgkamat, 2015-02-23:
@guyfleeman I got the garbage on debian, let me see if I can reproduce it now...

jgkamat, 2015-02-23:
[Here is the output of the garbage](http://pastebin.com/qxziHUDZ)... I printed out an ls before, for reference. I can get it to reproduce quite reliably actually (ive been working on my arch machine much more often)... Do you want me to bring my Debian machine over tomorrow, so you can take a look?

guyfleeman, 2015-02-24:
@jgkamat yeah whatever you think will help us find a solution. Maybe we can go through this tomorrow? I'm aware of the problem occurring when the mbed is plugged in, the OS queries some information and if you screen in too soon the reply is garbled and the OS can't communicate with it properly.

jgkamat, 2015-02-24:
@guyfleeman thats actually probably it! On my arch system, that is minimal (as there's no auto-mounting and such), but my debian system is running KDE, which does a ton of background work.

jgkamat, 2015-03-02:
This issue still affects me on debian, so there still must be another underlying problem besides the transfer of the file.

justbuchanan, 2015-03-03:
I haven't looked into it yet, but the reboot command that gets sent after the binary is copied isn't working (at least on my Arch system).

justbuchanan, 2015-03-10:
@jjones646, I mentioned the issue the other day about the MBED not rebooting after the `make robot2015-prog` command and you said something about sending a break command to the serial port.  Do you have anything to add to that?  It sounded like a better way of rebooting the MBED remotely than what we do now.

jjones646, 2015-03-10:
Setting a break condition on the serial line signals to the interface chip to perform a full reset. I haven't been able to make it work however. The [pyOCD](https://github.com/mbedmicro/pyOCD) library is the easiest thing I've come across. Load a binary that blinks some LEDs really fast to the mbed and run the python below while its connected to get an idea of what it does. It will load a binary file directly to the microcontroller's flash memory. Unfortunately, this takes orders of magnitude longer to do compared to pulling it from the on-board serial flash chip when needed at startup. You'll definitely be able to tell a difference if you flash it using the python library. This also doesn't load anything onto the USB flash storage area of the mbed. So there are a few downsides to it, but...there are ways to get around the long flashing time.

``` python
#!/usr/bin/env python

import time
from pyOCD.board import MbedBoard

board = MbedBoard.chooseBoard()
target = board.target
flash = board.flash

# Hault operations for 2 seconds
target.halt()
time.sleep(2)

# Resume operations for 2 seconds
target.resume()
time.sleep(2)

# Reset the mbed - This does NOT reload the LPC1768's memorie(s)
target.reset()
#target.resetStopOnReset

flash.flashBinary("Documents/robocup-software/build/firmware/robot2015/cpu/robot2015.bin")

# This will start whatever binary was loaded since it was placed directly into the LPC1768's flash memory
target.reset()
```

justbuchanan, 2015-03-11:
fixed in ecbfae93835fb492d5ace80ee174025040438453

closed 2015-03-11 by justbuchanan