RoboJackets/robocup-software #232: Setup travis-ci to build mbed firmware

justbuchanan, 2015-02-22:
Right now travis-ci gives us the nice green build indicator if the `make` and `make test-cpp` commands run successfully.  We should make the necessary changes so that it also runs `make robot2015` to see if the mbed firmware compiles successfully.

jgkamat, 2015-02-23:
I noticed that we are having a bunch of hacks to get travis builds working on ubuntu 12 (as they only support that). Maybe we could run the tests in an Ubuntu 14 docker instance, so we can both upgrade easily, and avoid having lots of special cases to get ubuntu 12 working...

justbuchanan, 2015-02-24:
Yeah that would be a pretty cool solution.  I've sorta been holding out hoping that travis-ci would upgrade to ubuntu 14 quickly, but it's been a while.

The main benefit of travis-ci is that it has good GitHub integration and reruns tests everytime you push to any branch.  I think it would be fairly difficult to get something similar setup with a custom solution.  If you have any ideas and have a good place to host docker images for free, that'd be awesome!

jgkamat, 2015-02-24:
I was thinking of just installing docker on the travis CI machines, there are some scripts to do that for us already...

jgkamat, 2015-02-24:
Actually, I just looked at the [install docs](https://docs.docker.com/installation/ubuntulinux/#ubuntu-precise-1204-lts-64-bit), and it looks like docker needs a linux kernel of 3.13. If the travis machines have the backported kernel, this could work, but it would likely fail otherwise...

justbuchanan, 2015-02-24:
Ahh darn, I liked the idea of loading a docker image up on travis.  I doubt they have the backported kernel.  I'm hoping they'll update to 14.04 soon and we can just run on a native travis VM without a problem.

closed 2015-02-25 by justbuchanan

jgkamat, 2015-02-27:
Ive been poking around, and it seems like [these guys](https://github.com/rocker-org/rocker/blob/56cb59cc262e9d2c02f9e11cfacab65f0b659c5f/.travis.yml#L10-L24) got docker working on travis CI. Im going to start trying to implement what they did on our builds!

justbuchanan, 2015-02-27:
Sounds good!  If we could get our CI builds faster and based on the regular ubuntu-setup script, that would be awesome.