RoboJackets/robocup-software #244: Implement one-touch-pass play

justbuchanan, 2015-03-06:
One play that a lot of teams use is to have a robot position itself a little ways away from the goal, then have another robot pass to it.  The stationed robot is positioned so that as soon as it receives the ball, it kicks it at the goal.  This year, our passing accuracy should be significantly improved, so this is something we should aim to have.  If we can get it to work in the simulator now, I think we can get the motion control to catch up.

Here's a video of us doing it a long time ago:

https://www.youtube.com/watch?v=U5mJghTbqKk

justbuchanan, 2015-03-25:
When you guys work on this, keep in mind that the code will be essentially the same as what we have now for CoordinatedPass.  The easiest way to implement it would probably be to pass a custom receiver class to CoordinatedPass, then we can have a PassReceiver, and a OneTouchPassReceiver.  That way the logic for setting up the pass, etc will stay the same.

justbuchanan, 2015-06-26:
Any update on this?  It would be really nice if we had this going for comp this year.

jgkamat, 2015-06-26:
I've mostly got this ironed out, minus the part where I decide where the receiver goes. I'm planning on just evaluating a field of points one by one, and choosing the one with the highest probability, does that sound like a good idea? To choose the points, I was thinking of manually creating proportional receiving zones, and picking the one opposite the side the ball is currently in, and dividing them up into points.

justbuchanan, 2015-06-26:
Sounds great!  Keep in mind that choosing the receive point should be the job of the play that this 'tactic' is used in, that way different plays can use it in different ways.  The code that looks at the field state and finds a good place to put the receiver to setup for this would go great as a module under the `evaluation` folder, then the play could be something like this:
~~~{.py}
## MyOneTouchPassPlay

one_touch_pass.receive_pos = evaluation.one_touch_pass_receiver_positioning.best_receive_point()
~~~

I do like the idea of specifying a fairly large rectangular zone, then evaluating several points inside it to find the best.  For each point you test, the "goodness score" could be as simple as evaluating the shot from that point to the goal and multiplying it by the evaluation of the pass from the ball's current position to that point.  We've already got functions for this so it wouldn't be too hard to do.  This way the "goodness score" is roughly proportional to the % chance that the pass and subsequent shot will both succeed and result in a goal.

jgkamat, 2015-06-27:
I got the evaluation bit and integrated it into onetouchpass (now on the touchpass branch). The evaluation is kind of flakey (in that it jumps all over the place), so I run it only once at the start of the play. I was thinking of running it at intervals later on in case anything changes, but is there a way to prevent the algorithm from jumping all over the field?

Also, the point-to-point evaluation picks passing points that are extremely close to robots, is there a way to put a buffer around obstacles for this particular evaluation?

justbuchanan, 2015-06-28:
Probably the best way to resolve the target point moving each time you run the evaluator is to only change the point if the new target is significantly better than the current one.  I think it does make sense to recalculate periodically (maybe every frame or every X frames), but each time you do, compare the score of the best position to the score of the current target.  Then you can have some sort of ReceivePointChangeThresehold that we could tune based on trying it out a few times.

I'll try out your stuff in a little bit and see what you mean by the passing points being too close to bots, so I'll get back to you on that one.

Are you talking about three points on the goal to try and shoot at or three points within the goal zone to position the receiver?

jgkamat, 2015-06-28:
The 3 points I was talking about were inside the goal, I was thinking about a center, left and right target within the goal. (I'm not sure how accurate we are if we aim off center).

I like the threshold idea, i'll play around with that and find a value that seems to work well!

justbuchanan, 2015-06-28:
Oh ok.  Keep in mind that we have a WindowEvaluator in the evaluation folder that's designed specifically for this purpose.  Rather than choosing a point on the goal line to aim at manually, just tell the window evaluator where you're shooting from and it will automatically pick the best point to aim for.  You can see it used in probably all of the plays/behaviors that we currently have that shoot at the goal.

When you're messing with the threshold, remember that if you change the value while soccer is running and save the file, soccer will detect the change and automatically reload the play/behavior from disk and restart it running, so you can really quickly try new values without having to restart things.  If you look at the terminal you launched soccer from, it'll print some log messages letting you know that it happened correctly.

Also, just a minor note, but if you use two pound signs for the first line of a python comment, it will show up in our doxygen comments on the docs website.  Keep the small comments within methods as one pound sign since they're more implementation specific, but it's nice to have class and method descriptions show up on the docs site.  See [here](http://robojackets.github.io/robocup-software/md_doc__documentation.html) for more info.

justbuchanan, 2015-06-28:
Try out the DebugWindowEvaluator play to see a good visualization on how the WindowEvaluator chooses its target point.

Also, I tried out the OneTouchPass play and it seems like it often chooses receive points that are on our end of the field when a point closer to their goal would be better.  Is this intentional or is this just what the output of eval_shot says to do?  If it's eval_shot, we may need to change it to prefer points closer to the goal.

jgkamat, 2015-06-28:
When I choose the rectangle, I make sure that it includes points with a lower y value than the ball, so if the ball is on the opponent's side of the field, it should work fine. This is because it would be impossible to bounce the pass at 100% kick speed, so right now, all onetouchpasses are passed backwards, and then bounced forwards into the goal.

PR #330 (2015-08-26): Add a onetouchpass play
This still needs a lot of cleanup, but I think its about time for this to be reviewed. Currently the only thing left to do is aim for areas that arent the center of the goal when using touchpasses (which is waiting on window evaluator improvements (that I could do), and possibly optimize my function for determining receive points (Right now it creates a list of points, and then iterates through them, we don't have to store this list).

This works best when the ball is on the enemy side of the field. Sometimes the ball won't replan if a robot gets in the way, this is bec [... cut]
closed 2015-08-26 by justbuchanan