RoboJackets/robocup-software #255: Improve Skill for kicking or capturing moving ball

justbuchanan, 2015-03-12:
A good resource to look at for this is RoboDragons ETDP from this year.

justbuchanan, 2015-08-23:
I was browsing TDPs and [RoboDragons ETDP](http://robocupssl.cpe.ku.ac.th/_media/robocup2015:tdp:robodragons.pdf) has a good description on how they do dynamic kicking - see section 3 "Dynamic Ball Kicking".

jgkamat, 2017-02-05:
@ashaw596 can we consider this closed, since you worked on this last year at comp, or does this need more work.

guyfleeman, 2018-06-04:
Is this mergeable with #1128?

ashaw596, 2018-06-04:
Its probably ok, but I don't think I had a good chance to test it with real robots. Not sure about current motion control, but it wasn't working too great last year at comp(didn't get too much testing tho with moving balls). It's mergable with #1128, but I'm not sure how much these changes would help depending on how well the robots actually move right now.

closed 2018-07-10 by Jason27chan