RoboJackets/robocup-software #256: Improve defense tactic

justbuchanan, 2015-03-12:
I started several improvements to the `Defense` tactic at comp last year, but didn't get around to merging them.  They're still on the `defense` branch and just need a little work.

Enhancements include:
* ignoring previous shell id for role assignment of defenders
* allowing a variable number of defenders rather than always having 2
* reducing the number of windowing calculations required per iteration

JNeiger, 2019-01-18:
@Alan-Penkar This may be of interest to you. I think you already have more than 2 robots and improving blocking of multiple threats.

I want to say pass estimation / shot score has been improved since this was made.

Alan-Penkar, 2019-01-18:
Thanks Joe, helpful to see the history- Defense.py should get broken out and rewritten by summer.  Since we're already discussing this and the issue is fairly vague, we can probably close this out though.

closed 2019-01-18 by JNeiger