RoboJackets/robocup-software #264: Replan just the last segment when there is a goal change

ashaw596, 2015-03-31:
Should be able to implement a very simple direct path for small goal changes. Add the plan onto the end of the path.

justbuchanan, 2015-03-31:
@ashaw596, when we talked about this yesterday, we discussed implementing this by adding some sort of function to `Path` for extracting a subpath.  One point you brought up was whether it should return a new `Path` object or modify the original.  I think it might be good to make it return a new one b/c it would make it easier to move path-planning to its own thread at some point.  We would leave the original untouched, then swap to the new path as soon as it's done being processed on the path-planning thread.

justbuchanan, 2015-06-30:
Any update on this one?

justbuchanan, 2015-06-30:
I guess we'll probably need the new path-planner in place for this to happen...  That part may not happen for comp this time around since there are several other things that take priority that need work (firmware mostly).

ashaw596, 2015-07-01:
Yeah it needs the new planner in place unless I fix the other planner. What
do we need to do with firmware? Or can I finish the new planner?

justbuchanan, 2015-07-01:
It'd be probably be best if I finish up the new planner since I've kind of got it in a weird state right now :/  Do you have an MBED on you by chance?

ashaw596, 2015-07-01:
=/ I didn't take one with me to cali.

ashaw596, 2015-07-01:
Lol I guess I could see if anyone in my company happens to have one. Probs
not tho.

jgkamat, 2015-07-01:
I'm pretty sure I brought one here, I could give it to you if you needed it!

justbuchanan, 2015-07-01:
Ok, there's probably not a whole lot you can do with the firmware then :/  How much work do you think it would be to setup the replanning to replan only part of the path if I am able to get the new planner in place in time?  If it's not too much, I'd say go ahead and do it, and hopefully I can get the planner done.

justbuchanan, 2015-07-01:
Yeah if you guys could arrange that, that would be awesome!  Not sure how far you guys are from each other.  I expect that the majority of our time at comp will be spent frantically trying to get firmware up and running, so it definitely wouldn't hurt to be familiar with it.

ashaw596, 2015-07-01:
Hey jay! Yea that be awesome. Can I grab that mbed sometime? Where do you
live?

jgkamat, 2015-07-01:
I'll send you an email!

justbuchanan, 2015-07-02:
@ashaw596, you were asking earlier about the parameters to the main planner method.  The planner isn't 100% finalized, but I don't think it'll change from this:
~~~{.cpp}
std::unique_ptr<Path> plan(const Geometry2d::Point& startPos, const Geometry2d::Point& startVel, const Geometry2d::Point& endPos, const Geometry2d::Point& endVel, const MotionConstraints& motionConstraints);
~~~

ashaw596, 2015-07-03:
@justbuchanan ok. Sounds good. Also can we refactoring motion constraints so that it makes more sense with our current planning system. The targetpos and targetworldvel don't make that much sense now that we're also setting a target velocity in for motion planning. I think it would be best to either make a seperate motion commands structure. Or create functions like setspeed or set pathplanning to make its use clearer and add a seperate path planning target vel and change the current one to like "overrideTargetVel" or something

ashaw596, 2015-07-03:
It's just weird the mix of both motion constraints and motion commands is confusing.

ashaw596, 2015-07-03:
Also, in the interface, ur probs missing the obstacles.

ashaw596, 2015-07-03:
Are we doing angle planning in the path btw?

justbuchanan, 2015-07-03:
Yeah I agree - let's split up MotionConstraints into one that's actual constraints (velocity and accel limits, etc) and one that's the command.  And good call, the set of obstacles will be in there as well.  The paths will ignore angles, I think it's probably best to handle that separately.

ashaw596, 2015-07-04:
btw, Can you help me with some boost python stuff? I'm trying to refactor the RObot method to use a unique_ptr to storethe path, but the boost python won't let it compile because it tries to export the RObot's copy method. Theres something like boost:noncopyable you can put in the class template, but I don't really get boost python @justbuchanan

justbuchanan, 2015-07-05:
Just downloaded it - I'll take a look and see what I can figure out.  I'm not sure why it needs the copy constructor though, we should only be passing pointers to Robots into python land, not making any copies.

justbuchanan, 2015-07-05:
Got it fixed!  You were right, adding boost::noncopyable fixed the problem.  By default, boost::python tries to make a copy constructor for the wrapped class, which in this case wasn't going to work.  Apparently we ran out of template arguments for the wrapper, so I removed the one for shared ptrs to OurRobot since we weren't using it and put noncopyable in its place.  It's on your albert/replan branch now.

PR #911 (2018-02-25): Fix line kick and forward pass.
This isn't really ready, but it should be pretty close after I get a chance to test with the robots.

TODO

- [ ] Test with real robots
- [ ] Fix up Ball prediction with real ball
- [ ] Clean up everything.

Closes #264
closed 2018-02-25 by jgkamat