RoboJackets/robocup-software #270: make robot2015-prog doesn't work on the field control computer

guyfleeman, 2015-04-13:
When running prog on the dual-screen Ubuntu machine by the field, the prog fails with the message "no mbed device found".

jgkamat, 2015-04-13:
I'll fix this on tuesday... Were you able to see the mbed in the devices list just fine? If so, something is going wrong in the name parser...

The ubuntu machine on the field is 14:04 right?

justbuchanan, 2015-04-13:
The field computer is on 14.10.

I ran ls before and after and the MBED showed up as "usb-mbed_Microcontroller_101000000000000000000002F7F148E9-0:0" under /dev/disk/by-id.

justbuchanan, 2015-04-13:
I did run this with a different MBED though than the one @guyfleeman used, so there's a chance that one has something messed up.

jgkamat, 2015-04-13:
Hmm, that device ID looks like it would be identified by the script fine, I dont see how that could cause the 'no mbeds found' error. I'll check out @guyfleeman's on tuesday...

jgkamat, 2015-04-15:
I tried it out, and it seemed to work fine. The git repo was not pulled in a while, and ubuntu-setup hadn't been run, so I did both of those, and everything seemed good. Try it out on thursday, and if it dosent work please tell me what shows up under ```/dev/disk/by_id```. If it works out, just close this issue! =)

justbuchanan, 2015-04-17:
@guyfleeman, when you get a chance, go ahead and merge the master branch into mbed-radio so you have the latest stuff - I think this will fix your issues.

justbuchanan, 2015-04-20:
I'm going to go ahead and close this - reopen it if you can reproduce the issue

closed 2015-04-20 by justbuchanan