RoboJackets/robocup-software #277: Nuke all tabs in favor of spaces

justbuchanan, 2015-04-15:


ashaw596, 2015-04-15:
RIP tabs.

joshhting, 2015-04-16:
Aww

jgkamat, 2015-04-16:
[This](http://nickgravgaard.com/elastic-tabstops/) is a pretty interesting concept regarding tabs. Unfortunately its not implemented yet

ashaw596, 2015-04-16:
inform supports it? lol

ashaw596, 2015-04-16:
There actually are some plugins for it.
https://github.com/SublimeText/ElasticTabstops

jgkamat, 2015-04-16:
Its kind of weird that plugin isn't on the main page. Some people are thinking about writing a vim patch for it too...

justbuchanan, 2015-04-19:
That is an interesting idea...  I actually like the way that works over regular tabs.  I'd rather not do anything though that requires everyone to have special editor plugins.

I'm closing this issue in favor of #161 since it encompasses spaces/tabs as well as all other formatting.

closed 2015-04-19 by justbuchanan