RoboJackets/robocup-software #278: Write firmware for MBED-based radio base station

justbuchanan, 2015-04-17:
A few things todo that I can think of right now:
- [ ] receive commands from `soccer` and broadcast them
- [ ] status LEDs - the on-board ones will be fine
- [ ] place received packets in a buffer to be read from `soccer`
- [ ] move MBED-related CMake stuff out of `robot2015` and into `firmware` so they can be accessed by both robot firmware and base station firmware
- [ ] change the `USBRadio` class in soccer to use the new base station

@jjones646 at some point it'd be good to get another radio board or two populated for developing this.

I think for this time around for radio stuff we should pretty much just copy the old radio protocol, then put our ambitious dual-channel base station etc. on the back-burner until everything else works.  The cool stuff will probably be a project for next year.

cc @guyfleeman

justbuchanan, 2015-04-17:
I wonder if we can change the USB vendor id and/or product id so we can detect the base station as a different device than robots

justbuchanan, 2015-06-28:
@jjones646, do you know if there's a good way that the computer can tell if a plugged in MBED is a base station or a robot?  Maybe something with changing USB product ids or something?

ashaw596, 2015-06-28:
Well there is always the lazy solution I wrote. You can put a file on the
device that only exists if the mbed is for use as a base station and check
for it when the mbed starts up.

On the other hand. I just realized you want something to tell the computer.
You could transmit the info over usb to the computer. But other than that
disregard this response.

barulicm, 2015-06-28:
Usually, projects use udev rules for this kind of thing. Key off of the serial number and link some to /dev/mbed-base

justbuchanan, 2015-06-28:
Sounds good to me.  I was thinking of using USB product id for the udev rules, but serial number sounds easier and just as effective.

jjones646, 2015-06-28:
If using the mbed USB libraries, you can set the `product_id` value from the constructor of the `USBHID` or `USBSerial` classes. Could also set the `vendor_id` value to an open USB vendor value for knowing when the mbed is running something other than base station / robot firmware binaries because the `vendor_id` would be the mbed's default value.

justbuchanan, 2016-04-04:
@guyfleeman I remember discussing serial vs HID for the connection a while ago, but I don't remember the details.  What are the pros/cons of going with HID?  I'm wondering if it might be easier to use serial since we won't need the extra usb connector.

guyfleeman, 2016-04-04:
@justbuchanan 

PROS:
From a reliability with re-connection standpoint I think HID is superior. I know the base doesn't necessarily need to be hot-pluggable but some serial quirks with this can be annoying, like if you write the line when linux is IDing the device, the output can get currupted. Also, for future events where we might need two base stations for 12 bots, I think it'd be easier. Of course, we might just put two 1201s on a base but I wanted to keep the option open.

LESSER PROS:
It would also allow us to keep our debugging channel open. Not exactly relevant since Ubuntu supports serial easily, but HID is more widely supported without a driver. HID libraries support non-blocking operations more easily, IMO.

CONS:
We need the libhidapi package(easy and small) and the external connector (less easy).  

OVERALL:
Both should work about the same, but HID is slightly more reliable and learning HID seemed like a good thing to do.

closed 2016-05-23 by justbuchanan