RoboJackets/robocup-software #279: Add verilog code to travis-ci

justbuchanan, 2015-04-17:
We can use icarus verilog for this - we don't need to test much, just see if it compiles successfully.

jgkamat, 2015-04-17:
Is this as simple as just running ```iverilog robocup.v``` in firmware/robot/fpga?

justbuchanan, 2015-04-17:
I think it might be actually.  @ChrisDoho, can you confirm?  Is there anything you had to do to specify a netlist or anything and did you have to compile the files separately?

Also, I think we'll leave the old fpga code alone and just test our new stuff, which will be in the robot2015 folder.  The new stuff still lives on the `mbed-fpga` branch and hasn't been merged into master yet, so we'll put the travis config changes there too.

ChrisDoho, 2015-04-17:
For simulating code in icarus, you do not need any .ucf files. You do need to make a "testbed" for the module you want to include. The test bed simulates the inputs and  writes outputs to specified .vcd file. You can look at any of the files ending in "_tb" in /robot2015/VerilogTest/ for an example.

If you just want to test if it compiles you don't even need the test bed, you just tell icarus to compile the .v files you have. I've never used icarus with a ucf file before so I don't know how it will handle it. As a heads up, Icarus is stupid about how it handles preprocessing. You have to run one command to create the preprocessed verilog files and another to compile those...

commit cf9716c5 (2015-04-19): added Verilog test to Travis. fixes #279
closed 2016-02-22 by justbuchanan