RoboJackets/robocup-software #288: Add method in Robot.hpp kickSpeed()

ashaw596, 2015-04-22:
Kicks the ball at the speed given in m/s.

justbuchanan, 2015-09-02:
Related to issue #422.

kylestach, 2019-10-02:
A new member will be working on this, so I've flushed it out a bit:

## Description
Currently, our kicking code allows the user to specify a power. This doesn't really have any physical meaning, and it's up to the user to tune it as they see fit. We'd prefer to allow the user to specify a "muzzle velocity" of the kick - that is, how fast the ball should be traveling as it leaves the robot - and the software should calculate that automatically.

The bulk of this project will involve actually figuring out a way to get an accurate estimate of the muzzle velocity from the robot. Most likely, the best way to do this is as follows:
 - Create a graph of the ball velocity during/immediately after a kick
 - Fit a line through the first part of the kick velocity (immediately after the kick) to get the intercept at the time that we actually kick the ball
 - Repeat for several different ball power values. Check with electrical to make sure that the values that you are using are safe.

Then, all of these recorded values can go into an interpolating map from speed to kick power.

Finally, we should update simulation to match what we see in real life.

## Requirements

- Create a mapping of kick power to kick velocity
- Create a function to interpolate between emperically-derived values
- In the kick function, use that derived to set the kick power
- Convert existing kick() calls python-side to use the new kick functionality.

## Design
[Link](#TODO)

closed 2024-02-01 by sid-parikh