RoboJackets/robocup-software #291: Use genetic algorithms for tuning motion control parameters

justbuchanan, 2015-04-23:


justbuchanan, 2015-04-23:
[This paper](http://arxiv.org/pdf/1204.0885.pdf) describes a similar thing with a PID controller.

jgkamat, 2017-01-27:
I want this to be a high priority item for Japan, as this is needed for per robot tuning (it would be a pain to do this by hand), and this will probably give us the biggest improvement to our robots at this point.

I think that hill climbing might be a simpler (and maybe better?) way to implement this, at least for PID, but I'm not sure (maybe we could use simulated annealing). Ideally this would be implemented mostly in common, and then we can use the same tuning code in firmawre and software (we can have robots auto-tune themselves without soccer sending any extra data). If we want to have soccer set firmware PID constants we would need to add some stuff to the radio (#304 above), but we could put all the PID tuning code in soccer, allowing it to be much bigger.

PR #887 (2017-07-20): Tune PID automatically
This will allow us to automatically tune per robot pid using a play. Closes #291.

- [x]  Make play to move back and forth along line
- [x]  Function to analyze the effectiveness of a certain set of pid values
- [x]  System to determine and test new pid values
- [x]  Find a way to have consistent robot movement for testing
closed 2017-07-19 by chachmu