RoboJackets/robocup-software #294: Re-examine rules related to entering the goal zones

justbuchanan, 2015-04-23:


jgkamat, 2015-04-27:
So, I've looked at [the rules](http://robocupssl.cpe.ku.ac.th/_media/rules:ssl-rules-2015-diff.pdf) again, and they say that a team cannot enter their opponent's defence area (and touch the ball). We cannot have any defenders touch the ball within our own area (other than the goalie).

See the bottom and top of page 26. The manual can't seem to decide between the spelling of 'defense' or 'defence'...

jgkamat, 2015-04-27:
I've noticed when messing around with my new play, if a robot is just over the edge, he will still go over the line (if the ball is on the edge) when making a pivot kick. The path planner stops the robots from going into the defence zones, but when in a play, there seems to be no restriction.

justbuchanan, 2015-04-27:
Sounds good - I guess we've got it setup now the way it needs to be.  The goal zone obstacles are created in the [GameplayModule's constructor](https://github.com/RoboJackets/robocup-software/blob/master/soccer/gameplay/GameplayModule.cpp#L80) and added to each robot's path planner depending on whether or not they're the goalie (for our goal zone).

The reason the robot goes over the edge is because the Capture behavior (used as the first stage of PivotKick) only uses the path planner for the course_approach state, then switches to velocity control for fine_approach.  If the ball is in range of fine_approach, the velocity control kicks in and we set the robot's velocity directly, which ignores any obstacles.  Eventually these will be unified and rather than setting velocity directly, we'll run that through the path planner eventually.  We'll leave this for now since it's an edge case that we don't see all that often and the path-planning isn't quite to the point where we can do this the way we want to.

closed 2015-04-27 by jgkamat