RoboJackets/robocup-software #298: Improve evaluation of shots and passes

justbuchanan, 2015-04-26:


justbuchanan, 2015-04-27:
One thing that we could use here that we don't do currently is model the % chance of kicking where we aim as a gaussian as described [here](https://books.google.com/books?id=oO6YNGtlzbcC&pg=PA298&lpg=PA298&dq=robocup+shot+gaussian&source=bl&ots=9xo_iq6A8I&sig=WKDCuM5wph57wJdpNzo_l1Z3vPA&hl=en&sa=X&ei=A6o9VZvjDZLOgwTi3oCYBA&ved=0CB4Q6AEwAA#v=onepage&q=robocup%20shot%20gaussian&f=false).  That's an old link, but the math is pretty timeless.  The basic idea is that when we aim for a certain shot angle, the angle the ball actual leaves the robot at will follow a normal distribution centered at the aimed-at angle.  The key here is to figure out the std dev of this distribution, which is something that we'll have to measure.  Once we know that, we can do a normalcdf using the angular width of the segment available to shot at, which gives us the chance that we'll make the shot.

ashaw596, 2015-04-27:
Once we figure out the variance and mean (I mean I hope it kicks straight
on avarage), boost seems to have a good implementation for this (assuming
it follows a normal distribution).
http://www.boost.org/doc/libs/1_42_0/libs/math/doc/sf_and_dist/html/math_toolkit/dist/dist_ref/dists/normal_dist.html

PR #915 (2017-03-12): Combine Window Eval and Kick Eval
Closes #298 #892 #902 
Would close PR #896 too

I found a closed form solution that should combine the two.
The idea is that there are 3 main probability / score distributions that need to be used to evaluate this.
1. Where the robot might be
2. What consists of a good kick
3. Our kick distribution

Distribution 1 is a set of normal distributions, 
Distribution 2 is a band pass filter looking distribution created using unit step functions
Distribution 3 is just a normal distribution.

1 and 2 are combined with addition to create a new set.
A convolution of 1/2 [... cut]
closed 2017-03-12 by JNeiger