RoboJackets/robocup-software #300: Investigate physics of ball and dribbler in simulator

justbuchanan, 2015-04-27:
Right now, the ball is always kicked straight from the robot, which doesn't match real life in the case of doing a touchpass.  The ball should be kicked at an angle depending on the direction that it comes to the robot at.

cc @jgkamat

jgkamat, 2015-04-27:
I tried decreasing the power of the receiver, and the kicker kicks at an angle, as expected.
[image]

Maybe this is only an issue when the kickpower is at max, or maybe when the kick power is at max, it's ratio is so high, it dosent matter. In any case, I'll work on finding the angle the robot should be, as the max kick power will be reduced at comp. It would be really helpful to have a expected kick speed function. Do you know how to get the current/predicted (at a point) speed of the ball?

ashaw596, 2015-04-27:
I think the predicted speed should be given by the BallFilter probably

justbuchanan, 2015-04-27:
Awesome!  I just assumed after seeing it the first time that the simulator was setup to always kick straight regardless of the ball's initial speed.

The kick speed at competition is limited to 8m/s, so when they do inspections, we tune a MaxKickPower config variable so that calling `robot.kick(1.0)` will kick slightly under 8m/s.  Ideally we'd have a `kickSpeed()` method where you could specify the desired kick velocity of the ball in m/s, but it doesn't exist yet (see issue #288).

The BallFilter calculates the estimated current position and velocity of the ball, which you can access in python with `main.ball().vel`.  We also have a [function](https://github.com/RoboJackets/robocup-software/blob/master/soccer/gameplay/evaluation/ball.py#L35) for predicting the position of the ball a fixed amount of time in the future given it's current state.  Right now it just predicts a future position, but we could add velocity as well.

I'll go ahead and close this issue for now if it's behaving as expected, but reopen it if you notice anything too weird.

closed 2015-04-27 by justbuchanan