RoboJackets/robocup-software #301: New Plays/Behavior Tree Panel dissapears when shrinking beyond a point

jgkamat, 2015-04-27:
As said, if you try to shrink the new right panel, it disappears.
[image]

Also, when shrinking the window beyond a point, the field overlaps the right panel.
[image]

Also, it would be nice if we could make the plays/behavior tree segments resizable, vertically. 

No major problems, but we should probably fix them before fall next year. I'll look into how qtcreator works over the summer

justbuchanan, 2015-04-27:
Just pushed a commit to master that puts the plays and behavior tree views in a vertical splitter so they're resizable.  I also prevented collapsing of the tab widget.  I'm not having the same issue with overlap on my computer, so maybe you can show me that on your computer sometime.

jgkamat, 2015-06-15:
Yah, this was only a problem on my messed up window manager, otherwise everything else is good now

closed 2015-06-15 by jgkamat

ashaw596, 2015-06-15:
=). Nice.