RoboJackets/robocup-software #304: Add parameter passing feature to radio protocol

justbuchanan, 2015-05-06:
For this year we'll be more or less copying the previous radio protocol since we're running low on time to implement some of the cool features we've talked about.  One thing we're looking at adding is a way to be able to change certain values on the robot at runtime.  This would be used for motion control parameters mostly, but would be flexible for other use cases.  @jjones646 and I like the way this works in the [crazyflie firmware](https://github.com/bitcraze/crazyflie-firmware/blob/56250b5ded22a25e8dc46eda95aa85bccd8e53b1/modules/interface/param.h).

justbuchanan, 2015-06-23:
This is obviously secondary to getting everything else working, but take a look at the crazyflie link above, @barulicm , when you get a chance.  @jjones646  showed this to me a while ago and I really like their parameter passing stuff they have setup with their radio protocol.  Adding something like this to ours would make motion tuning instantaneous rather than requiring a firmware reflash to tune a value.  Probably not something we'll get to this year, but good to keep in mind for the future.

jjones646, 2015-08-05:
I've been going through how they did this, so here's some things I've found out about how it is used in the crazyflie's system.

- Within the firmware, any variable that needs access is passed through a macro that stores the variable's type, memory address, and assigns a short name.
- There are 2 different macros: one for tagging a variable with read-only access, and one for tagging a variable with read/write access.
- Variables that relate closely with each other are added to a group by adding the variable macros in a struct. (There's macros for this too). This group is also set with a short name.
- During compilation, all variables within a variable group get placed into a known section of memory using a custom linker script. This happens because the macros add a variable attribute that specifies this.
- At startup, the base computer will request a table of contents. The crazyflie returns a string of all the group names with the variable names within each group (Don't know the exact specifics of this, but  that's the jist of it).
- The base computer then sends request packets that tell the crazyflie to return a variables value from table of contents at a specified time period (this would have to be changed). Variables are selected based on its index location in the table of contents.

I setup a linker script for the mbed that does this, but it doesn't work and I'm not sure why. I've been working on firmware stuff on my forked repo if anybody wants to look at it. The custom linker script is used with in build with the `robot2015-log` make target. I'll be cleaning up some things and send a pull request soon since so much is different from what's in any of the RJ branches.

justbuchanan, 2015-08-07:
Sounds good, I like their parameter feature a lot, it'd be cool to have.  How does the custom linker script work?  Is there a reason for doing it that way rather than just having a table in c++?

Also, make sure to merge master into your branch occasionally so you have the latest central changes.  I think there's a few things in there that are firmware-related that aren't on yours yet.

jjones646, 2015-08-07:
The linker script just defines where each compiled object should be placed
in memory. And outlines the placement for initialization in ram. My best
guess is that they do this to ensure that the variable can be accessed
since the macros are in the source file of what variable addresses it needs
to store. The script then makes sure the values are accessed straight from
the flash. The linker script I threw together seems to have issues with the
placement of object constructors/deconstructors.

Std should be able to handle this in some form or fashion I think. I've
looked over some of the std containers but haven't tried any of them for it
yet. Do you know of any boost libraries that would be worth looking into
for static const containers? It would be nice to have some macro that was
called in the same source file the variables are initialized in. Then the
param file would just put all of the macro calls into some static const
container at build time. The same setup would be useful for defining
console commands too.

I have a lot of merging to do, but I'll make sure the pull request contains
the most updated firmware for each section.

justbuchanan, 2015-08-10:
Ok makes sense.  I think you could get something similar to this in pure C/C++ without needing a linker script though.  It seems like you only really need to to the linker route if the param setting is handled at a really low level in the firmware.

Std containers can only hold a single type of object, so you can't intermix ints, floats, structs, etc in a single container.  What you can do though is store the addresses of variables.  If you want to address things by their index in the table, you could just use a std::vector<void*>.  If you wanted them keyed by names, you could use std::map<string, void*>.  The vector would be faster though.  You could do it something like this and then have a header file that's included in soccer and the firmware to define the indexes of variables.
~~~{.cpp}
typedef enum {
    ParamTypeFloat,
    ParamTypeInt
} ParamType;

typedef struct {
    std::string name;
    ParamType typecode;
    void* address = nullptr;
} ParamEntry;

std::vector<ParamEntry> GlobalParamTable;
GlobalParamTable.reserve(100); // allocate 100 entries

#define REGISTER_PARAM(index, name, typecode, address) GlobalParamTable[index] = {index, typecode, address};
~~~

Yeah merge conflicts are pretty inevitable haha.  Just keep in mind that merging master into your branch often is usually much easier than trying to merge a lot of stuff later.  I changed a good bit of CMake stuff so it shares as much as possible for the base station and control board firmware build setup and also added support for unit tests.  I don't think that's on your branch yet, which will be a fun merge if you've changed much in CMake.

jjones646, 2015-09-03:
I found a potential solution: https://developer.mbed.org/cookbook/RPC-Interface-Library

jjones646, 2015-09-06:
I got it working over the serial interface! The hardest part about the mbed's RPC stuff was integrating it into the build system. It's an official mbed library, but doesn't look like the `build.py` script supports building it. So I just added it as a library in the cmake stuff, and made it a dependency for the `common2015` library.

I added a command called `rpc` into the console that takes whatever is given to it and passes it straight to an RPC call. So all of the RPC commands that are documented on the mbed website should work for it.

Here's a screen shot showing 1) getting a list of variables/objects that can be called on, 2) reading one axis of the accel, and 3) reading/writing a variable in the code. I just pushed all of this to the `robot2015-firmware` branch.

[image]

justbuchanan, 2015-09-07:
Awesome, nice work!!  Being able to read and write things remotely will open up a lot of possibilities.  How tough do you think it'll be to make this work over the radio?

jjones646, 2015-09-08:
It will be pretty straight forward. The main concern of radio packet timings arises because the values will reset on a power cycle unless they're written to a file or something similar.

closed 2018-07-01 by guyfleeman