RoboJackets/robocup-software #305: Double check kick-power limits

barulicm, 2015-05-13:
Tests today showed kick powers above 255 and two distinct levels for 0 and 7e-15

jjones646, 2015-06-25:
I started at the lowest kicking power and ended at the highest values from using the manual controller.

Here's what soccer showed:
0 (no kick)
7.07E-15 (no kick)
25.5
51
76.5
102
127.5
153
178.5
204
229.5
255
280.5 (the end result from this value was a kick that was slightly less powerful than that from 25.5)

ashaw596, 2015-06-26:
Btw, if you collected speed data for this, could we use it for creating a power to speed conversion? #288

closed 2015-06-27 by barulicm

jjones646, 2015-06-27:
@ashaw596 You could, but I don't know how beneficial it would be without a lot of testing and experimenting. The relationships between applied power and final ball velocity is pretty weird. I can tell you that from looking at everything when I did this, there's very little difference for power levels between 127.5 and 255 for the new solenoid. This will need to be adjusted and normalized in the new firmware.

commit 79333dd6 (2015-06-27): Fixes joystick kick power and dribbler speed limits. Closes #305