RoboJackets/robocup-software #307: Add the ability to have text layers in soccer default to hidden

justbuchanan, 2015-06-17:
When drawing shapes or text on the soccer field view, we pass a string indicating the "layer" that these show up on.  The "layers" tab has a check box for each layer to show or hide the content.  Right now every layer defaults to being enabled, which can lead to a lot of unneeded debug content being drawn that clutters up our pretty soccer field view.  It would be nice if we could add a way to hide some by default, then we can enable them when we're debugging specific things.

justbuchanan, 2015-06-17:
cc @ashaw596

ashaw596, 2015-06-17:
Fiiiinnnneee. After the other one has been merged tho lol.

justbuchanan, 2015-06-17:
Awesome!  Let me know what your thoughts are on how to do this.  Not sure at the moment where it would fit best.

justbuchanan, 2015-06-30:
Hey @ashaw596, I'd like to use this feature when I finish #254.  Do you think you'll be able to get to this soon or should I look into it?

ashaw596, 2015-06-30:
I'll take a look at it.

ashaw596, 2015-06-30:
Ok. I've changed it to hide all layers by default to start. What exactly is the functionality that want? @justbuchanan

justbuchanan, 2015-06-30:
Well there are many layers that we *do* want to have on by default, there are just some that are better to start off hidden.  Maybe we can have a method setLayerDefaultVisibility(QString, bool), then we can call it with the appropriate layers in the constructor.  We just need make a clear note of this functionality in the header.

ashaw596, 2015-06-30:
Do you know which ones we want by default. Well just have to create them when the program starts.

justbuchanan, 2015-06-30:
If someone adds a new layer, I think it'd be confusing for it to be hidden when soccer runs.  It'd be nice if we can just set visibility to false for the ones we know right now that we don't usually want to see.

For now, I think just the path-planning related ones like "reusing path", etc should be hidden.  Another one I'd like to hide by default is the ssl-vision packet data, but I'll add that later.

ashaw596, 2015-06-30:
Well, most of them get dynamically created. So they'll need to be instantiated separately if we want different use cases. I'll make a seperate create function for the debug layers later tonight.

justbuchanan, 2015-06-30:
What if we just have a map of default visibilities:
~~~{.cpp}
std::map<QString, bool> _defaultVisibilities;
~~~

This will only be read the *first* time an item is drawn to a layer - everything after that just goes by whether or not it's checked in the UI.

ashaw596, 2015-06-30:
=/ I don't know if that makes sense. I think it would be easier just to convert this to use a constructor type method. We can use a list to construct it if we want it to be easy to read.

justbuchanan, 2015-06-30:
Not quite sure what you mean... What constructor are you talking about changing?

ashaw596, 2015-06-30:
Well I wanted to make it less strangely decoupled with the really weird creation mechanisms since blocking isn't terribly important since it only happens every once in a while. I refactored part of it in fieldView and Mainwindow, but gave up on the SystemState/logFrame system. I added the hiddenList String array to MainWindow.cpp.

PR #328 (2015-07-05): Albert/hide layers
Maybe what you wanted? It hides the motion planning stuff at least.
Closes #307
commit 2133944c (2015-07-05): Merge pull request #328 from RoboJackets/albert/hideLayers

Creates a list of debug layers which are disabled by default. Closes #307
closed 2015-07-05 by ashaw596