RoboJackets/robocup-software #309: Infinite Loop Style

barulicm, 2015-06-18:
Three options, pick one:
- while(true) { }
- while(1) { }
- for(;;) { }

We really should be consistent. I'm not intimately familiar with arm-specific compilers, but normal gcc compiles these down to the same instructions. I'm partial to while(true) as the clearest, but it doesn't matter as long as we don't switch every time we write a loop.

ashaw596, 2015-06-18:
I think true is good. Or 1

justbuchanan, 2015-06-18:
I agree.  I personally think the first is most readable, but as long as they're consistent.

barulicm, 2015-06-19:
mbed has no 'true' / 'false' constants? I can use a smart pointer, but can't write "while (true)" what language is this? So disappointed :frowning:

justbuchanan, 2015-06-20:
Are you in a .c or .cpp file?  I see bools all over the place in robot2015 and it builds fine on my computer.

barulicm, 2015-06-20:
I think I actually had something wrong with my build setup. I was in main.cpp having problems, but they seem to have disappeared now. It may also have had something to do with a few unnecessary extern "C"s that I got rid of.

commit 9d2b22e4 (2015-06-28): while (1) -> while (true).  closes #309
closed 2015-06-28 by justbuchanan