RoboJackets/robocup-software #313: Firmware logging

justbuchanan, 2015-06-22:
One thing I've liked a lot about the logging utilities at Google is that calls to `log()` use an ostream rather than printf, which makes it easier to log arbitrary data types.  An example usage would be if we wanted to print out an Eigen::Vector2f:

```{.cpp}
Eigen::Vector2f vec(1.5, 2.5);
log(INFO) << "My vector = " << vec;
```

This is a little harder to do with printf-style logging since you'd have to manually call a toString function on whatever it is you want to log.  The toString function still has to exist, obviously, you just don't have to explicitly call it when logging.  The method for the vector would be something like this, which is included with the Eigen library:

```{.cpp}
ostream &operator<<(ostream &stream, const Eigen::Vector2f &vec) {
    stream << "(" << vec.x() << ", " << vec.y() << ")";
}
```

Any thoughts @guyfleeman, @barulicm?

barulicm, 2015-06-22:
I like it. I was already thinking of cleaning up the logging code this week. ROS has similar support for stream-style logging which I've found convenient while working on other projects this year.

justbuchanan, 2015-06-22:
Sounds good.  I'll try and send a PR for this tomorrow after work sometime.

I was thinking about also cutting down the log levels from:
* FATAL
* SEVERE
* WARN
* OK
* INF1
* INF2
* INF3

To just these for simplicity:
* ERROR
* WARN
* INFO

barulicm, 2015-06-22:
Agreed. I don't think that many levels will actually be helpful.

justbuchanan, 2015-06-22:
Actually, Matt, if you're familiar with the stream-based logging in ROS, do you want to go ahead and do this?

barulicm, 2015-06-22:
Sure, I can take care of it.

justbuchanan, 2015-06-23:
When you get around to this, let's put it onto master so I can use it on non-radio firmware things.

closed 2015-06-23 by barulicm