RoboJackets/robocup-software #317: Store CMake ExternalProjects locally?

justbuchanan, 2015-06-26:
Not sure how our internet connectivity is going to be this year, but last year it was fairly shitty :/.  Several of our dependencies right now are pulled from online repos into the build folder when running `make`, which is pretty awesome in general and doesn't need to happen more than once, but things have to be re-downloaded after a `make clean`, which might be a problem...  It wouldn't be hard to just copy those repos into a third_party folder in our repo, then point our CMake ExternalProjects to the local copies to avoid having to be online.

jgkamat, 2015-06-27:
We should probably fork/create a permanent separate branch for this, and work it in a way so that the repo size dosent blow up permanently.

justbuchanan, 2015-06-27:
When you do a git clone or fetch, it downloads the content on all of the branches, so branching it wouldn't actually save us any space here.  We could fork the repo for it, but I don't expect it to more than maybe 20MB, so I think it's probably ok.  If we really wanted to delete it form the history, we could rebase master after comp and delete the commit, but it probably isn't worth the effort.

barulicm, 2015-06-28:
Are the external dependencies hosted via git? If so, I'd recommend using git's submodules feature.

justbuchanan, 2015-06-28:
Yeah that's a good point, submodules would probably go well here.  Unfortunately a couple dependencies are through mercurial, so I guess we could mirror them as git repos on our account or just copy the files into the third_party folder.

commit 8b0ac01c (2015-07-01): moved external dependencies into git submodules.  closes #317
closed 2015-07-01 by justbuchanan