RoboJackets/robocup-software #320: Add --timeout flag to simulator

justbuchanan, 2015-06-28:
This way we can optionally have the simulator exit after a few seconds of not receiving any messages from soccer.  See PR #319.

jgkamat, 2017-01-13:
This is really bugging me, and I would love it to be fixed by someone new! I don't think it would be _too_hard, and it would definetly give you a nice overview of protobuf (possibly) and C++, although it would deal with our simulator code, which may eventually get deprecated.

jgkamat, 2017-01-15:
Example of using rj::time https://github.com/RoboJackets/robocup-software/blob/d18e9b7fd56f42e1888e494064e0c264a9cceae8/soccer/Timeout.hpp

Place to potentially add reset timer: https://github.com/RoboJackets/robocup-software/blob/d18e9b7fd56f42e1888e494064e0c264a9cceae8/simulator/physics/Environment.cpp#L351

this is the main method: https://github.com/RoboJackets/robocup-software/blob/d18e9b7fd56f42e1888e494064e0c264a9cceae8/simulator/simulator.cpp

kevinatorchen, 2017-01-15:
How to exit out of the simulator correctly?

jgkamat, 2017-01-15:
This stackoverflow post has a little bit of info on it :http://stackoverflow.com/questions/8026101/correct-way-to-quit-a-qt-program

From what I can tell, there should be a `quit()` method that should do what we want.

In the main method, there is a 'QApplication' object, which you could call `exit()` on as well.

I'm not sure if these work though, so test them out and let me know what happens!

PR #872 (2017-02-05): Timeout
Closes #320
closed 2017-02-05 by jgkamat