RoboJackets/robocup-software #321: Investigate using protobufs or something similar for radio communication

justbuchanan, 2015-06-28:
Google protobuf or some of the even more light-weight variants (capnproto and mayb eothers) might be a good choice of binary format for us for radio communication in the future.  Currently we use protobuf for radio messages internally (RadioTx and RadioRx), but we flatten it down it down as much as possible before sending to the robot.  This is nice for speed, but protobufs add a lot more flexibility with only a slight bit of overhead.  I haven't looked into it enough, but the little bit of extra overhead might be worth it to make our lives way easier in adding some of the new features we want, especially now that we have a faster on-board processor and faster radios.

jjones646, 2015-07-27:
@justbuchanan What's the general process of setting this up for the cross compilation? I really like this idea and from what I've seen online about google protobuf, the reduced efficiency shouldn't really be an issue if the new motion control stuff works well.

justbuchanan, 2015-07-29:
It's actually pretty easy.  The protobuf compiler, `protoc` reads the .proto file and spits out the corresponding .cpp and .h files, which are the same for the MBED and host computer.  After that, the .cpp files are compiled with the right toolchain for the target (MBED or host).  See [this file](https://github.com/robojackets/robocup-software/blob/comp2015/common/protobuf/CMakeLists.txt#L10) for an example.  I can set it up when it comes time if you want.

One thing to keep in mind is that repeated and optional fields change the size of the output depending on their values, which will affect the timing of our protocol.

Also, protobuf has overhead when parsing messages, but there are third-party variants (capnproto for example) who's accessor methods just index into the message at read time, which might be better for our use case.

jjones646, 2015-11-25:
I've checked out [Cap'n Proto](https://capnproto.org/index.html) a little bit, and it looks pretty legit. The bit ["packing"](https://capnproto.org/encoding.html#packing) feature being the main point that could be taken advantage of. It's highly focused on encoding long streams of zeros.

For our purposes, every long stream of zeros in a packet would be reduced to 2 bytes. There is some overhead for using cap'n proto obviously, but this might make better use of the airspace in the long run. Still a little early to say that without actually seeing what a radio packet would look like with cap'n proto, but it does look promising. Especially if you could somehow figure out how to sort and place all motor velocities together for the base station's tx packet (~34 byte reduction when robots aren't moving, ~28 with 1 moving. Kinda a long shot to implement, but interesting nonetheless.).

With a synchronous protocol, this would make things more complex. In the sense that smaller packets (average) doesn't equal more packets. There would just be empty airspace until the start of the next radio frame since that's set from the worst-case timings for everything that's communicating. Unless the base station did something like dynamically determine when the start of the new frame would be.

justbuchanan, 2015-11-25:
I've done a little messing around with capnproto and its encoding format [here](https://github.com/RoboJackets/robocup-software/tree/justin/radio-protocol-brainstorm-2/firmware/common2015/modules/CommProtocol).  If you check out that branch and cd into that folder, then run `make encode`, it will encode the message in `example.txt` with and without packing and tell you how many bytes each requires.  It makes it easy to see what type of overhead we'll be getting with capnproto.

I wonder if we can get a variable timeslot setup going where each robot replies in order after the previous one is done.  The base station would still determine the order of robot replies, but it would be up to the robots how much they send.  In competition mode, they'd typically send a minimal status back to the base station, but at other times, they could do more (like when we're syncing parameters, transferring files, etc).  I think it would make it easier to implement the ports-based communication you started working on.

justbuchanan, 2016-02-03:
Closing this.  We'll be going with byte-stuffing in C++ since protobufs have a lot of overhead :cry:

closed 2016-02-03 by justbuchanan