RoboJackets/robocup-software #333: Add license information

justbuchanan, 2015-07-09:
RoboCup SSL has a yearly open-source trophy that we will be competing for.  In order to submit our codebase to be considered, we need to have a clearly-stated license.  I'll do some research, but we'll most likely go with MIT.

jgkamat, 2015-07-09:
+1 on using the GPL. I don't mind MIT though :smile:

jgkamat, 2015-07-09:
also, github made [this website](http://choosealicense.com/licenses/) comparing the most popular licenses.

justbuchanan, 2015-07-10:
Cool, thanks for the link!  I like the easy-to-read breakdown.

I'm personally more of a fan of MIT for this because of GPL's copyleft-ness.  Our software really isn't anything groundbreaking, it's just a bunch of semi-decent implementations of known solutions packaged together to work well for RoboCup.  If someone liked a component of our stuff, such as path-planning, motion-controls, etc and wanted to take it and modify it for their closed-source app, I'd be cool with it.  Also, our code might be a good starting point for a beginner team and I'd hate for them to avoid using it as a starting point for the reason that they'd be forced to open-source their codebase for the rest of eternity.  If we get to the point that our code is state of the art and sharing it is a threat to our winning at every comp, it might be best to move to a model where we publish all our changes as open-source after competition is over, but keep our improvements for the year private until then.  However, I think that's quite a ways off :)

I think that given the current state of our software, we can only stand to benefit from having more people use it and potentially push bug-fixes and improvements upstream and I think GPL would prevent many potential users from considering it.  I'm open to other thoughts on this though if you think GPL would be a better choice for us.

jgkamat, 2015-07-10:
The GPL dosen't force other teams to open-source their entire codebase, for example, if I forked bash and made a really cool improvement, I don't have to open source that improvement if I keep it to myself. However, if I give it to someone else and they ask for the diff in the source, I have to give it to them. Teams can still use our modifications internally, but if they give their program to anyone to use, they must give the source when they ask. I only prefer the GPL because I feel like if someone gets our source (as a result of us open-sourcing it), then they should return that favor to the community as well (if they decide to distribute it). 

BUT: I'm perfectly fine with the ASL or the MIT licenses, and I would much rather have us be properly open-source than with no license (or a broken one like the truecrypt license). :stuck_out_tongue:

justbuchanan, 2015-07-10:
Ok, I guess you're right on that.  I think I'd still like to leave open the possibility for someone to use one a component of our stuff in their closed source program that they distribute if they want to although obviously we'd love to have them send us any improvements :)

justbuchanan, 2015-07-13:
After doing some reading, I think Apache is slightly better for us here.  I'll add some license info in a bit.

commit 686b73d5 (2015-07-13): added Apache v2.0 license. closes #333
commit 43735f2a (2015-07-13): added Apache v2.0 license. closes #333
closed 2015-07-13 by justbuchanan