RoboJackets/robocup-software #334: Pull field dimensions from vision packet

barulicm, 2015-07-17:
SSL-Vision now includes field geometry in vision packets. We should at least add the option of pulling field constants from this data. This will help smooth over the pains of the inevitably erroneous fields at competition.

justbuchanan, 2015-11-22:
Here are the protobuf files for ssl vision: https://github.com/RoboCup-SSL/ssl-vision/tree/master/src/shared/proto

And here's the part of our code where we read the vision packets: https://github.com/RoboJackets/robocup-software/blob/master/soccer/Processor.cpp#L298

Also take a look at #335

jgkamat, 2016-01-17:
Are these packets implemented in our simulator? I tried out a basic setup and I'm getting 0 for width/height.
https://github.com/RoboJackets/robocup-software/commit/592739285db0eab2823f383b01eac7dcb3df5efd

barulicm, 2016-01-17:
The simulator sends packets, but it might not fill out all of the fields. So they're atleast partially implemented.

PR #544 (2016-04-19): Add resizing of soccer based off of the geometry vision packets (if they are filled out)
I think I've gotten most of the kinks worked out, but I haven't tested this out in any way whatsoever. I'm going to try it out on the field and maybe try to get grsim working with it. This is just what I have so far.

Closes #334
closed 2016-04-19 by jgkamat