RoboJackets/robocup-software #343: Use config system for damped joystick speeds rather than hard-coded values increase them

justbuchanan, 2015-07-17:


justbuchanan, 2015-07-17:
These are at the top of Processor.cpp right now.

justbuchanan, 2015-11-12:
Damping can be toggled on/off in soccer via a checkbox in the "Manual" menu item in the top bar.

Changing our current hard-coding to configurable values that can be adjusted in the "Config" tab in soccer requires a few things:
* Add `static ConfigDouble` items for each damping value we want to control.  These will be class variables in the `Joystick` class.
* Declare a static `createConfiguration()` method in Joystick.hpp, then implement it in Joystick.cpp - this initializes the config values and specifies their name/keypath in the config system.
* register `Joystick` with the config system by adding a call to `REGISTER_CONFIGURABLE()` in Joystick.cpp

You can look at the [WindowEvaluator](https://github.com/RoboJackets/robocup-software/blob/master/soccer/WindowEvaluator.hpp) class for an example of how this is setup for a different class.  Also take a look at [Configuration.hpp](https://github.com/RoboJackets/robocup-software/blob/master/soccer/Configuration.hpp) where our config system is defined.

Offboss, 2015-11-14:
From what I can see there is no Joystick.cpp class. There is a Joystick.hpp, but no associated .cpp. However, there are two sets of other files that might fit the bill (GamepadJoystick and SpaceNavJoystick). Should I put the config stuff in both, or just Joystick.hpp, or other?

justbuchanan, 2015-11-14:
Forgot there wasn't already a Joystick.cpp, but we can go ahead and make one.  For this we want the damping values (max speed, max accel, etc) to be the same regardless of the joystick type (3d mouse or gamepad), so I think the Joystick class would be a good place to put them since it's the superclass of the two.

When you add Joystick.cpp, you'll need to add an entry to the `ROBOCUP_LIB_SRC` variable in the [soccer/CMakeLists.txt](https://github.com/RoboJackets/robocup-software/blob/master/soccer/CMakeLists.txt#L8) file.

Offboss, 2015-11-14:
Everything seems to be working, I will have to test if it actually works on Tuesday but all the adjustable values are present in the config in soccer so I am hopeful. I'll just comment and clean now.

PR #525 (2016-01-31): added config system for normal and damped joystick speeds
Added the speed of damped and normal joystick translation and rotation to the config system. Closes #343
commit 979ef07f (2016-02-12): Merge pull request #540 from jcarn/configurableJoystick

Adds configurable values for robot speed and rotation, closes #343
commit c3a384d0 (2016-02-12): Adds configurable values for robot speed and rotation, closes #343
closed 2016-01-31 by Offboss