RoboJackets/robocup-software #360: When the ball is invalid, its comet trail ghosts to (0, 0)

justbuchanan, 2015-07-19:


ashaw596, 2016-02-07:
Did this ever get fixed?

jgkamat, 2016-09-12:
Nope, this is still an issue. To reproduce, drag the ball off the field.

rk1998, 2017-01-11:
Where can I get started with this issue? Which files should I be looking at?

jgkamat, 2017-01-11:
I can still reproduce this, so this is a good issue!

I would start looking right here: https://github.com/RoboJackets/robocup-software/blob/33aafebf5ce594e604b2df3fdcda6e41c6f99f77/soccer/FieldView.cpp#L268

This section (right below it) is where we actually draw the ball trail (commenting it out should remove the trail completley): https://github.com/RoboJackets/robocup-software/blob/33aafebf5ce594e604b2df3fdcda6e41c6f99f77/soccer/FieldView.cpp#L281

The logic seems right for me, so you'll need to play around with it a bit and see what you find. Let me know if you're stuck!

I would also reference the QT documentation since we use QT in those functions a lot, such as:
http://doc.qt.io/qt-4.8/qpainterpath.html
http://doc.qt.io/qt-4.8/qpointf.html

PR #879 (2017-01-27): Fix comet trail ghosting to (0, 0)
 I added a bool value to make sure the pos.moveTo() runs within the for loop. Now the trail does not ghost to (0,0) when the ball goes invalid.

closes #360
closed 2017-01-27 by jgkamat