RoboJackets/robocup-software #366: Find a good way to run test plays without defense

justbuchanan, 2015-07-21:
Right now we have to comment out lines of code to make this happen

jcarn, 2016-03-02:
I'd be interested in this, which files does this involve?

jgkamat, 2016-03-03:
This would be a semi-major restructure, but doing it would help you learn a lot about the play framework.
[this is our basic offense play](https://github.com/RoboJackets/robocup-software/blob/33a3b43f5f9b04c72b234fd3a1b473a800990f21/soccer/gameplay/plays/offense/basic_122.py). [This is where the current defense plays are added](https://github.com/RoboJackets/robocup-software/blob/33a3b43f5f9b04c72b234fd3a1b473a800990f21/soccer/gameplay/plays/offense/basic_122.py#L52). Currently we add the defense tactic to plays that need defense, like shown on that line. We probably should find a way to have plays explicitly request defense from the assigner (and the assigner could choose whether to fulfill that request or not).

jcarn, 2016-04-07:
Are there any plays that we want to ignore the UI telling it to Use/Not Use defense?

justbuchanan, 2016-04-07:
There are a couple test plays that don't really make sense with defense disabled, but I think when we want to run those, we can just make sure the defense checkbox is checked.

PR #630 (2016-04-18): Created standard_play.py and made all plays functional and inheriting standard_play.py
From what I can tell everything works like it should, but I would appreciate if people with more experience could verify that the plays are running normally. I'll continue checking in the meantime. 

I created new class standard_play. It inherits from play.py. Right now, the only thing it does is handle defense for every play based on the status of the "Use Defense" checkbox in the UI. For every play that inherited play.py, In the future, any action that a normal (but perhaps not every play) play should impl [... cut]
closed 2016-04-18 by justbuchanan