RoboJackets/robocup-software #368: Fix role assignments for defense

justbuchanan, 2015-07-21:
The defenders switch places when threat priorities change.

justbuchanan, 2015-07-29:
This happens because sometimes the straight-line distance from robots to target points produces a different ordering than it should.  The solution is to use the estimated travel time instead of straightline distance when calculating assignment cost.  This is tricky though because travel time estimates can be more expensive to calculate...

ashaw596, 2015-07-29:
I think it would still be worth it to explore how this type of "estimate"
to be exposed through the path planner api. It would be helpful for a lot
of problems like this

justbuchanan, 2015-07-29:
One thing to keep in mind though is that right now role assignment runs at 60Hz and this estimate would be run (# robots)*(num roles that specify a position) times per iteration.  The path planner is expensive to run and even if we had a nice way to calculate a path and estimate travel time, we probably wouldn't want to use it here for performance reasons.  I think rough estimates should do fine here, so maybe just using the estimate from a trapezoidal velocity profile would suffice.

justbuchanan, 2015-07-29:
On second thought though, the assignment issue with defenders happens because the estimate ignores obstacles, so the trapezoidal motion profile approach wouldn't solve this...

ashaw596, 2015-07-29:
If you want to try trapezoidal estimates it is exposed now in the
robocup-py file

ashaw596, 2015-07-29:
Though yeah that wouldn't be any different from nearest robot really

closed 2022-07-12 by kasohrab