RoboJackets/robocup-software #369: Don't attempt to score right away on indirect kicks

justbuchanan, 2015-07-21:
Right now we do, which would count against us if we were able to make it.  The ball must touch another robot before we can score.

jgkamat, 2016-06-13:
I think I can do this. It seems that both indirect and direct kicks are handled by OurFreeKick (as far as I can tell). How would I check to see if we are doing an indirect kick or not?

justbuchanan, 2016-06-13:
Sounds good.  I don't know all of the details offhand, but there's a section on indirect kicks in the rules that has the info: http://wiki.robocup.org/images/1/18/Small_Size_League_-_Rules_2016.pdf.

jgkamat, 2016-06-16:
oh, I was asking about finding it more programmatically, do we have it in the c++ side somewhere yet? I would assume its sent to us from the ref.

justbuchanan, 2016-06-16:
Something like [this](https://github.com/robojackets/robocup-software/blob/master/soccer/GameState.hpp#L85)?

jgkamat, 2016-06-16:
Perfect, thanks!

I probably should have looked in that file before

commit 8a64856f (2016-06-18): Improve behavior on indirect_kicks

Make coordinated_pass have basic compatibility with any _kick.py object
Use coordinated_pass w/ a line kick when doing indirect kickoffs
Closes #369
commit 98bfb13a (2016-08-02): Improve behavior on indirect_kicks

Make coordinated_pass have basic compatibility with any _kick.py object
Use coordinated_pass w/ a line kick when doing indirect kickoffs
Closes #369
closed 2016-08-02 by jgkamat