RoboJackets/robocup-software #394: Fix neopixel assembly code for building with gcc

jjones646, 2015-08-25:
Someone may know of a better solution, but gcc doesn't seem to support some of the ARM assembly commands. So those commands have to be replaced with their equivalent opcodes for the thunb2 instruction set I believe.

jjones646, 2015-09-10:
This will compile now. It ended up being just syntax issues, so the assembly instructions themselves are the same with the exception of the register list for the push/pop instructions. A runtime error occurs however.
I have gotten a neopixel to light up to a specific color by forcing a reset, so the core of it actually works.
Haven't really looked into what's going on with it, but its likely that the assembly code losses track of the return address and never returns from the function. This would make since because of the push/pop assembly instruction changes moving that value around. Another possibility is it just not playing well with rtos.

justbuchanan, 2015-11-04:
@ryanstrat, what's the status on this?

jjones646, 2015-11-04:
@justbuchanan I was able to get this integrated into the cmake files. It was all syntax changes basically. However, running it in hardware causes it to get stuck in the nop loop I believe. Either resulting from the assembly lines changes for the push/pop instructions that gcc required, or something to do with not setting the RTOS up right before entering the assembly code when it runs.

jjones646, 2015-12-19:
[Done!](https://goo.gl/photos/G322VVNuvg4GiuqK9)

closed 2015-12-19 by jjones646