RoboJackets/robocup-software #399: Create basic class to create variables from rotary selector

jjones646, 2015-08-26:
See pin configurations [here](https://github.com/RoboJackets/robocup-software/blob/robot2015-firmware/firmware/robot2015/src-ctrl/config/pins-ctrl-2015.hpp).

mosdragon, 2015-11-04:
Hey @jjones646, I'm looking to pick this up. I was going to create a function that simply gets the int value of based on the pins that are on:
```c++
// The robot's rotary ID selector
#define RJ_HEX_SWITCH_B0        RJ_IOEXP_A0
#define RJ_HEX_SWITCH_B1        RJ_IOEXP_A1
#define RJ_HEX_SWITCH_B2        RJ_IOEXP_A2
#define RJ_HEX_SWITCH_B3        RJ_IOEXP_A3
```

Is this all what we're looking for?

mosdragon, 2015-11-04:
Actual OOP class, constructor takes in location of pins, and ```int read()``` would give us the hex value between 0x0 and 0xF from those pins?

Justin mentioned interfacing with the I/O expander to get the values of these bits? Any suggestions for which other files to look at to move forward with this?

jjones646, 2015-11-04:
@mosdragon all of the files [here](https://github.com/RoboJackets/robocup-software/tree/robot2015-firmware/firmware/robot2015/src-ctrl/drivers/mcp23017) is pretty much everything. Yes, the defines you commented on earlier are correct. All of the defines in that block need some kind of function/method to interface with whatever it corresponds with. This area on the control board actually has an error in the design causing it to not work, but I will get you something either Sunday or Tuesday so you can run any code on the actual I/O expander.

justbuchanan, 2015-11-12:
One idea that would be useful for this would be to template the rotary selector class on the type of pin interface it uses.  This will let us use the same code to interface with a selector connected directly to the MBED pins (using MBED's `DigitalIn` class) as well as one connected through an IO Expander (using our `IOExpanderDigitalInOut` class).  I can show how the template stuff will work if you're not familiar with it.

The main benefit to this is that you can wire up a selector directly to the MBED and test it out without having an IOExpander, but have it work just the same on the final board when it's connected through the expander.

justbuchanan, 2016-02-14:
Any progress on this, @mosdragon?

mosdragon, 2016-02-15:
@justbuchanan Sorry man, I've been swamped down past few weeks. This Tuesday I'll come in and have it working

justbuchanan, 2016-02-15:
Yeah no problem, I understand.  And sounds great!

justbuchanan, 2016-03-17:
@mosdragon I may go ahead and finish this one up so we can cross it off the list.  Do you know what the current status is?

mosdragon, 2016-03-17:
Hey @justbuchanan sure thing! I meant to tell you earlier, I probably won't be able to make it to meetings now because of my workload.

https://github.com/mosdragon/robocup-software/tree/firmware2015/rotarySelector

I never got around to testing it like you said to do, but it now compiles.

justbuchanan, 2016-03-17:
Sounds great, I'll take a look at it!  And no worries, Tech will do that to you haha

PR #577 (2016-03-30): Rotary selector class for robot shell id
This should be about ready to go, just needs to be tested on hardware.  Closes #399.

cc @mosdragon @jjones646
closed 2016-03-30 by justbuchanan