RoboJackets/robocup-software #400: Enable support for interrupts on the digital IO expander

justbuchanan, 2015-08-26:
The io expander (mcp23017) has 8 inputs and 8 outputs.  We need to set up a system to interrupt the mbed when one of the input lines change.

cc @jjones646

justbuchanan, 2016-02-03:
@jjones646 What's the status on this?

justbuchanan, 2016-03-28:
@jjones646 is this something that we'll need or should I close this?

jjones646, 2016-04-04:
We'll need this.

justbuchanan, 2016-04-04:
How would something like this work?  It's connected via i2c, so what line(s) serve as the interrupt?

jjones646, 2016-04-04:
There's another line for the interrupt itself - should be defined in the pin declarations file. It's a fairly trivial thing to do, but the lack of testable hardware is definitely the burden here.

jgkamat, 2017-07-01:
Closing due to timeout (and that this is firmware, not robocup-software).

closed 2017-07-01 by jgkamat