RoboJackets/robocup-software #403: Add a code coverage badge to the readme

justbuchanan, 2015-08-26:
This would require using a tool to generate a C++ coverage report, then adding something to our circle config to run generate the coverage and upload it to coveralls.io, which is where we'd pull the badge from.  See the [stylize repo](https://github.com/justbuchanan/stylize) for an example of how this is setup for python code.

This probably isn't worth doing now, but might be something cool for the future.  It'd be neat to see the coverage number go up when you add tested code to the project :smile:

jgkamat, 2015-10-08:
Congrats, we have a whopping 0.56% code coverage. That's including firmware though, I don't know if I should be excluding it.
[coveralls woo](https://coveralls.io/builds/3782732)

jgkamat, 2015-10-08:
Actually it seems to be including the mbed libraries and a bunch of other stuff, so its probably better than that. I'll keep tweaking this...

jgkamat, 2015-10-08:
Not really sure how this is working out, now it says 8%? 
https://coveralls.io/builds/3782766

justbuchanan, 2015-10-08:
Hahaha nice!  I have a feeling we're above 0.56% but below 8% although that's just a guess.  On the coveralls site it looks like it's including generated cpp files.  I see common/git_version.cpp and soccer/qrc_main_icons.cpp which are both generated at compile time.  Somehow they both have 100% test coverage too, which is interesting...  Maybe there's a way to exclude all files under the /build directory?

jgkamat, 2015-10-08:
Right now I'm telling it to scan all files under build, isn't that where our compiled files are? I think it wants the object files. I'm trying to play with the input root dirs, but /build is one that seems to work. I'll keep messing around with it. It seems to be including the googletest files in the output, so theres definitely something wrong...

justbuchanan, 2015-10-08:
My bad, I forgot it needs to look at the object files and not the source.  Haha testing the testing framework is some pretty advanced cs.  Looks like the tricky part will be getting all the excludes right so it doesn't count googletest and the many things that the qt moc spits out.

commit ea4847f6 (2015-10-08): Add a coveralls badge to readme
Closes #403
commit 496f631e (2015-10-18): Add a coveralls badge to readme
Closes #403
commit 58ec0484 (2015-10-27): Add a coveralls badge to readme
Closes #403
commit 17e2a0e6 (2015-10-27): Add a coveralls badge to readme
Closes #403
closed 2015-11-01 by jgkamat