RoboJackets/robocup-software #406: Travis-ci should only run on the master branch

justbuchanan, 2015-08-26:
We have this configured in our .travis.yml, but it doesn't seem to be working - travis runs on all branches.  I've opened a ticket with the travis team, so we'll see what they say.

https://github.com/travis-ci/travis-ci/issues/4708

jgkamat, 2015-08-26:
See [this pr](https://github.com/RoboJackets/robocup-software/pull/401)
[the travis build for this pr](https://travis-ci.org/RoboJackets/robocup-software/builds/77264330)
[the travis yml at the commit hash of the pull](https://github.com/jgkamat/robocup-software/blob/4ee0d5af7f864ca3bd17e6389618e24006c0839f/.travis.yml)

The tests are actually running?

justbuchanan, 2015-08-26:
Good point, I updated the other thread with that info.  It seems like it may be an issue just with pull requests, although it still shows a status for non-master branches that aren't PRs, which seems a little counter-intuitive.

justbuchanan, 2015-08-27:
It turns out that travis-ci builds pull-requests even when only the master branch is whitelisted.  I disabled pull request builds in their web console, so this is resolved.

closed 2015-08-27 by justbuchanan