RoboJackets/robocup-software #409: Solve Versioning Annoyances and/or provide support for Alternative Linux Distributions

jgkamat, 2015-08-26:
Over the last year, we keep having to work around 'we need package x but package x is not on 14.04', but the whole point of using an LTS distribution is to provide stability by not upgrading packages. If we keep adding repositories, we turn ubuntu 14.04 into a franken-pseudo-rolling distro, and no one likes those (and they break in bad, bad, bad ways).

We can solve this problem in a couple of different ways:

1. Switch to an easy to use rolling distro such as debian testing/unstable or opensuse tumbleweed.
2. Use docker to build, but not to run. In this case, we can choose the latest ubuntu environment to compile our program, while running it on any computer (assuming no dependencies are required to run run soccer). Compile docker-engine to see how it works. (this is different from nubot's system, which is painful to develop in). This could probably be made in conjunction with normal builds.
3. Update to the latest ubuntu every 2 or so releases (this could break people's systems).
4. Use nix, a package manager that works on any distro. It has newish packages, but has the advantage that any nix command will work on any system. I don't like this idea too much.

Please comment with suggestions/alternatives!

I'm still thinking about this, but it would be nice not to add sketchy symlinks and ultimately support any new distro. 

Overall, most software projects don't have complicated setup scripts and such, and still support multiple distros. We should work to have an experience closer to these projects.

jgkamat, 2016-12-13:
Once we get fedora we should be good IMO, and then I'll go ahead and close this issue. Manually setting up RC isn't too hard.

[image]

jpfeltracco, 2016-12-14:
Wait, so are we switching to officially support Fedora only now?

jgkamat, 2016-12-14:
not really official support, but I will be building a list of packages that /should/ work on fedora. Probably much the same scenario as what we have for arch, in the end.

jgkamat, 2017-01-10:
In the interest of keeping things generic, I'm going to write a docs page on how to set up rc-software on any linux distribution, instead of just fedora. All of our dependencies are pretty common so it shouldn't be too bad.

closed 2019-08-25 by kylestach