RoboJackets/robocup-software #412: Convert all the math.h imports to <cmath

ashaw596, 2015-08-30:
replace all 
```
#include <math.h>
```
with 
```
#include <cmath>
```

yousifd, 2015-10-05:
I made the modifications, how do I push the files?

jgkamat, 2015-10-05:
Cool, nice job!
I noticed you havent forked this repo yet, so here's what you need to do:
1. Fork the repo, this will create a copy of the repo in your own namespace. There's a button to do this in the top right of this page.
2. Go to a terminal and `cd` into the dir that you made the changes to
3. Copy your fork git repo and paste it into the command below. Based off of your username, your command should be:
`git remote set-url origin https://github.com/yousifd/robocup-software.git`
4. Run: `git add -A`
5. Run: `git push origin master`
6. Open a pull request here. There should be a little green icon next to the 'branch master' on your repo to help you do this. 

If you get stuck just comment here, and I'll help you out

closed 2015-10-05 by justbuchanan