RoboJackets/robocup-software #414: Refactor motion commands

justbuchanan, 2015-08-30:
As discussed at the meeting, we'll have something like this:

~~~{.cpp}

enum class AngleCommandType {
    FacePoint,
    FaceGlobalAngle,
    None
};

struct AngleCommand {
    AngleCommandType type();

    float maxSpeed;
    float maxAccel;
};

enum class MotionCommandType {
    TargetPath,
    TargetPos,
    TargetVel,
    // CaptureBall,
    None
};

struct MotionCommand {
    MotionCommandType type();

    /// Default values come from config values
    float maxSpeed;
    float maxAccel;
};

~~~

justbuchanan, 2015-08-30:
@ashaw596, do you think it would make sense to have several subclasses of `MotionCommand` - one for each type?  We can either do that or have the one `MotionCommand` class store all the data that any of the other types would need such as:
* MotionInstant - position target
* Path - for setting path directly
* float - target velocity

ashaw596, 2015-08-31:
I don't know if subclasses make much sense. The only thing they have in common would probably be getCommandType.

ashaw596, 2015-08-31:
Actually yeah maybe subclass would be best. lets see.

ashaw596, 2015-09-01:
Looking at using double dispatch pattern instead  of enum stuff.

closed 2015-09-14 by justbuchanan