RoboJackets/robocup-software #418: Move the robot status panel into the tab widget

justbuchanan, 2015-08-31:
Once our new robots are in good shape, we shouldn't have hardware issues nearly as often as we do now.  It would give us a lot more room to expand the FieldView if the robot status panel was placed into the tab widget and not always visible.  We should also make the tab turn red whenever a robot has a "showstopper" so we can still easily see if there's a problem.

jgkamat, 2015-08-31:
We should probably make 'minor issues' like hall faults turn the tab light orange, as we don't need to worry about them during a match.

justbuchanan, 2015-08-31:
That's true, although we're hoping to never see hall faults again with the new fleet (fingers crossed)!

closed 2024-02-01 by sid-parikh