RoboJackets/robocup-software #431: Improve path visualization with colors

justbuchanan, 2015-09-02:
Right now we draw the paths robots are following, but it's hard to tell how fast.  If we draw the path in red for max speed, blue for zero speed, and blend in-between, I think we can get a pretty nifty visual.

cc @ashaw596

jgkamat, 2015-09-02:
Maybe we should keep the trails the same color as the team (keep yellow as yellow) but increase the brightness/saturation of the color depending on speed. (eg. fast yellow robots have a bright yellow trail, but slow yellow robots have a white-yellow or muted yellow trail)

justbuchanan, 2015-09-02:
Yeah that's a neat idea!  For this, I was thinking more about the path we draw (pink/red line from the robot to where it's going) rather than the comet trail.

jgkamat, 2015-09-02:
ooh that makes sense, that's probably a pretty important thing for debugging the velocity profiles too.

commit 1de65eac (2015-09-08): Robot paths are now drawn in red for high speed and blue for low speed. closes #431

* Currently only applies to InterpolatedPath. other subclasses will have
  to implemente it as well
closed 2015-09-08 by justbuchanan